tino/pyFirmata #37: Read/write I2C

zawilliams, 2015-03-26:
Is there a way to read and write over I2C? I may be totally missing it.

tino, 2015-03-26:
Not directly, but I2C message are just sysex messages (https://github.com/firmata/protocol/blob/master/i2c.md), so you should be able to do something like:

``my_board.send_sysex(pyfirmata.I2C_REQUEST, my_data)``

You also want to write a method that does something with the reply. Subclass Board and add a command handler for the I2C_REPLY command:

```
class MyBoard(pyfirmata.Board):
    def _handle_i2c(self, *data):
        # do your thing

# instantiate and register the cmd handler
my_board = MyBoard()
my_board.add_cmd_handler(pyfirmata.I2C_REPLY, my_board._handle_i2c)
# use my_board
```

zawilliams, 2015-03-26:
Got it - thought that was the case. Thanks @tino!

closed 2015-03-26 by zawilliams