tino/pyFirmata #39: Digital Reads Return `None`

jochasinga, 2015-07-17:
When I did this on an Arduino Uno:

```python
>>> board.digital[3].mode = INPUT
>>> board.digltal[3].enable_reporting()  # Not sure why this has to be an explicit method call
>>> print(board.digital[3].read())            # Return None most of the time
```
The last statement return `None` most of the time. It did work sometime, returning `True` on an open switch and `False` on closed. However, I had to explicitly start a thread with

```python
>>> thread = util.Iterator(board)
>>> thread.start()
```
And with that it still returned `None` very often.

tino, 2015-08-02:
It is not necessary to call ``enable_reporting``, that is taken care of when you set the mode (https://github.com/tino/pyFirmata/blob/master/pyfirmata/pyfirmata.py#L473). 

You do need to run the Iterator, or call ``board.iterator()`` yourself, otherwise the values are not updated from the serial data.

Do you still get `None` values after non-`None` values when you run the iterator? That should not be the case.

jochasinga, 2015-08-02:
Hi @tino. After I ran board.iterator I got boolean for "high" and "low"
half the time. Still got None half the time.

Is it possible to implement `multiprocessing` instead of `threading.Thread`
for Iterator class? Threads in Python can be problematic due to GIL and I
am suspecting that.

Panisuan Chasinga
Interaction Designer
Edlab, Teachers College
Columbia University
525 West 120th Street
New York, NY 10027

m: +1(347)609-2075

[image: My LinkedIn] <http://www.linkedin.com/in/jochasinga/>  [image: My
Github] <http://github.com/jochasinga>  [image: My Twitter  ]
<http://twitter.com/jochasinga>

tino, 2015-08-02:
I am not that familiar with multiprocessing. And the thread used to work
fine for me (I have no current arduino project to test). If you can whip up
a multiprocessing solution that works better I am happy to add it as an
alternative to utils.py.

pinocchiodrc, 2015-08-06:
Hi, I have also made pin3 as INPUT, but it cant refresh. Someone knows the Problem? 
```
board.digital[3].mode = pyfirmata.INPUT
board.digital[3].read()
´´´ 
It returns every time True, no matter what I did. Do I need a Iterator? Or whats wrong with my code?
thanks

tino, 2015-08-06:
Yes, you do need an iterator, otherwise your board's values will not be
updated, and eventually your serial will overflow.

jochasinga, 2015-08-06:
@pinocchiodrc you'll need to start an `Iterator` thread before reading
```python

>>> board.digital[3].mode = pyfirmata.INPUT
>>> it = util.Iterator(board)
>>> it.start()
>>> board.digital[3].read()
```

Most of the time it work, but sometime `None` still show up. I'm looking into this part also.

pinocchiodrc, 2015-08-06:
@jochasinga  Thank you, I just tried, this time comes "pyfirmata is not defined". I guess the syntax is wrong. So I tried
```
pin3 = board.get_pin('d:3:i')
it = util.Iterator(board)
it.start()
board.digital[3].read()
```
but comes "None". I just begin to learn python and wanna write a GUI to control Arduino. This thing really drives me mad.

jochasinga, 2015-08-06:
@pinocchiodrc first, you got *pyfirmata is not defined* because you did not import it properly.
```python
>>> import pyfirmata                        # if you did not import the module
>>> board = Board([port_name])
>>> board.digital[3].mode = pyfirmata.INPUT # here you will get *pyfirmata not defined*
>>> it = util.Iterator(board)
>>> it.start()
>>> board.digital[3].read()                 # True
```
You might want to check if:
+ You have connected a digital input actuator to your board (i.e. a push button) to test.
+ When the button is pushed, see if `Tx` light (if you're on an Uno) is flickered. 
+ Change pin number.

Switching to `board.get_pin('d:3:i')` is the same as calling `board.digital[3]` so that shouldn't be the case. Getting `None` means it's a serial issue.

tino, 2015-08-06:
@jochasinga  is right, you have to import the pyfirmata module first.

Btw, ``get_pin`` is a utility method that sets the mode, and enables reporting when necessary. So you can do: 
```
>>> import pyfirmata                        # if you did not import the module
>>> board = pyfirmata.Board([port_name])
>>> it = pyfirmata.util.Iterator(board)
>>> it.start()
>>> pin3 = board.get_pin('d:3:p')
>>> pin3.read()
```

But we are getting a little off track.

@jochasinga, can you maybe check/log what the `_handle_digital_message` (https://github.com/tino/pyFirmata/blob/master/pyfirmata/pyfirmata.py#L346) is receiving? Cause after being set to True or False for the first time, a digital pin can not be set back to `None` (see https://github.com/tino/pyFirmata/blob/master/pyfirmata/pyfirmata.py#L434).

pinocchiodrc, 2015-08-07:
Hi, today I had another try. This time it works. But the pin state is not stable. 
```
...
board.digital[8].mode=pyfirmata.INPUT
board.digital[9].mode=pyfirmata.INPUT
board.digital[10].mode=pyfirmata.INPUT
board.digital[11].mode=pyfirmata.INPUT
board.digital[12].mode=pyfirmata.INPUT
board.digital[13].mode=pyfirmata.INPUT
it = pyfirmata.util.Iterator(board)
it.start()
def PrintPin():
     for i in range(8, 14):
         print('Pin', 'i', board.digital[i].read())
...
```
the result comes sometimes True and sometimes False, no matter what I did, only PIn13 works well. I guess there is something wrong with Iterator. Or something happend to the resistor in my Board.

tino, 2015-08-07:
Do you have something connected to those ports (ground or a high pin)? Otherwise the ports will "float" and will randomly switch. This has nothing to do with the iterator, this also happens if you check the value in arduino code.

pinocchiodrc, 2015-08-07:
@tino thanks a lot, I unterstand now. I should have a 1k resistor with pin connected.

jochasinga, 2015-08-08:
@tino This is what I got from `_handle_digital_message`:
True (open/button not pushed) => port_nr: 0, lsb: 8, msb: 0 
False (closed/button pushed) => port_nr: 0, lsb: 0, msb: 0

I still got `None` today for a while until after I got a [Coolterm](http://www.macupdate.com/app/mac/31352/coolterm/download) connected on that port to read the serial then `read()` started to return `True` or `False` again.

Sekhmet, 2016-02-21:
Hi,
I'm using pyFirmata 1.0.3 under Windows and I get None all the time.

```python
import pyfirmata, sys
from pyfirmata import Arduino, util

board = Arduino('COM7')
board.digital[8].mode = pyfirmata.INPUT
it = util.Iterator(board)
it.start()
print board.digital[8].read()
sys.stdout.flush()
```

I tried to implement every solution in this issue but I still got None here.

Port 8 is being connected to 5V with 10k resistor. FirmataTest detects it as high without any problems.

tino, 2016-10-29:
@jochasinga does the delay as described above work for you?

jochasinga, 2016-10-30:
@tino well that makes sense. I'm closing this issue for now unless someone reopen again.

closed 2016-10-30 by jochasinga

Vladivarius, 2018-01-18:
I also had this problem and solved it by adding some delay after starting the iterator thread. Some time is probably needed for the serial communication to be initialized propertly. 10ms is enough.

board = Arduino('COM5')
BTN = board.get_pin('d:10:i')
thread = util.Iterator(board)
thread.start()
sleep(0.01)
print(BTN.read())