tino/pyFirmata #43: Blink an LED with DIGITAL_PULSE 0x91

pinocchiodrc, 2015-07-28:
first of all thanks for the nice library.
Im just trying to setup an LED blink function with standardFirmata. But I didnt find the Sysex in Firmata.h. 
I did so, but I think it wont work.

`#define DIGITAL_PULSE 0x91`

`void Digitalpulsecallback() `
`{`
 ` digitalWrite(13, LOW);`
 ` delayMicroseconds(1000);    `
  `digitalWrite(13, HIGH);`
  `delayMicroseconds(1);    `
  `digitalWrite(13, LOW);`
  `delayMicroseconds(1000);   `
`}`

`void setup()`
`{`
 `Firmata.attach(DIGITAL_PULSE, Digitalpulsecallback);`
`}`

Is there someone, who knows how to make this and what should I in standardFirmata add? Thanks!

tino, 2015-08-02:
I am not sure what your question is. Did you try this? Does it work? Does it give you errors?

pinocchiodrc, 2015-08-06:
sorry, I mean I didnt find the command. Now I know there is actully no Pulse_SEQUENCE command in Firmata. I creat a new by myself and it works now.
thanks!

closed 2015-08-06 by tino