tino/pyFirmata #50: pin write and python sleep() issue?

tanthua, 2016-11-11:
Hi tino,

First I want to say thank you for your work. Really appreciate that. I have a question regarding the Python 3 time sleep() function. My set up is consisted of the Arduino UNO running Firmata firmware and a PC/laptop running python code using pyfirmata library.

What I am trying to do is to analog write to a PWM pin from a list of values (100 values in this case). I also want them to output in every 0.001 second or 1 ms. Therefore, technically, 100 values in 0.001s/each would take ~0.1 sec to complete.

Then the problem happened when I used timer to see how long it would take. It only achieved ~0.1 sec total when Google Chrome was opened on certain websites (I know it seems weird, but it's true 100%). Even when I minimized Google Chrome, the total time would slow down to 1.6s consistently until I opened Chrome again.

Then I tried to take off my time.sleep(0.001) in my loop then it worked fine regardless the appearance of Chrome or not.

I really want to understand why and a solution for this case. Maybe an alternative of time.sleep? Here is my code in python. You can load the firmata sketch from Arduino IDE.

Also note at higher value of time.sleep(), everything works fine with or without Chrome opened.
`
import pyfirmata
import time
from timeit import default_timer as timer

board = pyfirmata.Arduino("COM4", baudrate = 115200)
pwmPin = board.get_pin('d:5:p') #digital pin 5 pwm mode

my_points = [1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84,85,86,87,88,89,90,91,92,93,94,95,96,97,98,99,100]

def LoadWaveform():
   rate = 1/1000
   #print (my_points)
   while True:
      start = timer()
      for i in range(len(my_points)):
         pwmPin.write(my_points[i])
         #time.sleep(rate)

      end = timer()
      print ("Total time run: ", end - start)
      time.sleep(1)

LoadWaveform()
`

tino, 2016-11-12:
I am not completely sure what problem is. You seem to have it working when you don't put in a `time.sleep()` right?

Google Chrome being of influence on the sleep or execution time seems highly unlikely, unless it's eating all your CPU.

Also be aware that if you run this on Python 2, `1/1000 == 0`.

tanthua, 2016-11-12:
Thank you for the response. It does work fine without time.sleep(). I was able to reproduce the problem by putting the block of codes into a second thread, and it slowed down regardless of what is opened. Thinking of how to fix that now because my application needs to send data constantly and I don't want it to be interrupted by other apps.
Again, thanks a lot.

tino, 2016-11-18:
Hmm, yeah processes will fight for CPU time. Get a bigger one ;).

Note that pyFirmata is great for prototyping and getting stuff working. If you some high performance or more stability, it often pays of to write some stuff that runs directly on the Pi (and thus is not affected by Chrome and the likes). The waveform function is easily ported to C++, and you can set it up as a sysex function. That way you can still call it from python, but it runs directly on the Pi.

Good luck!

closed 2016-11-18 by tino

tanthua, 2016-11-18:
I finally got it working. It's the OS system. Not pyfirmata :D

tino, 2016-11-21:
Ah sweet!