tino/pyFirmata #57: TypeError: unorderable types: NoneType() > int()

unique1o1, 2017-01-28:
i'm trying to turn OFF and ON an LED according to the data from a LDR using this code: 
   ` from pyfirmata import Arduino, util
board=Arduino('/dev/ttyACM1')
led=board.get_pin('d:13:0')
ldr=board.analog[0]
it = util.Iterator(board)
it.start()
board.analog[0].enable_reporting()
x=board.analog[0].read()
if x>20:
	led.write(0)
else:
	led.write(1)

`

and it gives me this error
`TypeError: unorderable types: NoneType() > int()`

whereas using the `board.analog[0].read()` in a python shell give a float.
What am i doing wrong here?

Monstamac, 2018-02-04:
I have the same issue. The type comes up as <type 'NoneType'> and I struggle to cast it to anything.

tino, 2018-04-16:
@unique1o1 I suspect the error is thrown on the `if x>20:` line?

Use `if x and x > 20:` instead. `read()` can return `None` in the beginning when there aren't any reads yet.

closed 2018-04-16 by tino

JervisNW, 2021-09-27:
Try print(x) after x=board.analog[0].read()

I had the same problem and that is how a solved it.

from pyfirmata import ArduinoMega, util, pyfirmata
import time

LED = 3
arduino = ArduinoMega("COM3")

#arduino.analog[7].mode = pyfirmata.INPUT  
it = util.Iterator(arduino)
it.start()
time.sleep(0.05)
arduino.analog[7].enable_reporting()
i=0
while(True):
  sensor=arduino.analog[7].read()
  print(sensor)
  if i>100 and sensor > 0.500:
    arduino.digital[LED].write(1)
  else:
    arduino.digital[LED].write(0)
  i=i+1