tino/pyFirmata #65: board.getpin( ).write( ) | ArduinoMega2560 problem

hydridity, 2018-02-22:
Hi, i'm developer for team school project where we use servo motors, and i found your firmata client as most suitable for us but i'm having problems with sending commands.
As in your example stands 
```
import pyfirmata
board = pyfirmata.ArduinoMega('PORT')

pin = board.get_pin('parameters')

pin.write(value)
```
im trying to use DIGITAL I/O pins (22 - 54) with parameter 'd:22:s' in get_pin but there is my problem that it won't do anything after passing pin.write
only case where it works for me is when i set servo pins at arduino PWM pins (2-14), parameter 'd:2:s' 
but when i tried firmata test software setting servo at digital pins 22 - 54 works like expected
im trying to find where is problem but i can't find out if it's known bug or there is something that i doing wrong

and also i noticed at the start the function write() of class Pin is called with value 0 that cause painful flick to servo motors

hydridity, 2018-04-08:
Found, problem, but not solution yet, 
as there are values transfered under ANALOG MESSAGE command, like in Pin class 
its write function `msg = bytearray([ANALOG_MESSAGE + self.pin_number, value % 127, value >> 7])` 
for both, Servo and PWN pins, well on arduino mega there are 15 PWM pins, this analog message will be transmitted with pin_number more that 15, but firmata analogCallback attached function can recognize it only if pin_number is less than 15
using these PWM pins can cause problems when somebody using almost all internal Arduino Timers.
If it was on purpose then i'm just saying that all digital pins (at least on arduino mega) can be used for servo PWM generation as servo library using same internal timers

soulsynapse, 2023-07-14:
Running into the same problem 5 years later, pretty much got exactly as far as you did. No fixes on my end, just consolidating pins which is what I should have done to begin with.