tino/pyFirmata #72: Setting pin to INPUT_PULLUP

void4, 2018-09-01:
It seems to be implemented in the Firmata library, does pyfirmata support it as well?

dimasalifdianugrah, 2019-04-17:
it seems until now still not yes supported. hopefully it soon as possible

RazowSy, 2019-05-21:
still no update for pull up?

ale-novo, 2021-05-13:
Hi, i ve added INPUT_PULLUP in my fork:
https://github.com/ale-novo/pyFirmata

commit:
https://github.com/ale-novo/pyFirmata/commit/65bc649e579f9883f60992da06184907df5af007

you should use pin mode as INPUT_PULLUP

denuradhan, 2021-06-09:
> Hi, i ve added INPUT_PULLUP in my fork:
> https://github.com/ale-novo/pyFirmata
> 
> commit:
> [ale-novo@65bc649](https://github.com/ale-novo/pyFirmata/commit/65bc649e579f9883f60992da06184907df5af007)
> 
> you should use pin mode as INPUT_PULLUP

its work's thank you

SkyDaz233, 2021-12-18:
> 

It doesn't seems to work for me tho, i'm using this syntax

COM = board.digital[9]
COM.mode = pyfirmata.INPUT_PULLUP

SkyDaz233, 2021-12-18:
> Hi, i ve added INPUT_PULLUP in my fork: https://github.com/ale-novo/pyFirmata
> 
> commit: [ale-novo@65bc649](https://github.com/ale-novo/pyFirmata/commit/65bc649e579f9883f60992da06184907df5af007)
> 
> you should use pin mode as INPUT_PULLUP

It doesn't seems to work for me tho, i'm using this syntax

COM = board.digital[9]
COM.mode = pyfirmata.INPUT_PULLUP