tino/pyFirmata #74: read analog pins

Kushal4092, 2018-09-10:
I have simple code which reads A0 and i have my DMM next to me which says its 0.002V and pyfirmata gives me None
0.1955
0.1848
0.1848
0.1828
0.1828
0.2991
0.3226
0.3255

when I Multiply with 5v I dont even get the value which is on the DMM. So let me know if you what I should do to fix it. the Values are inconsistent. Basically there is not voltage at that pin but and times I get random values. I know that for 0 to 5v its 0-1 in pyfirmata.

ozkansen, 2020-09-23:
I encountered this problem too. Is there any solution?

Kushal4092, 2020-09-23:
May I look at you code.

ozkansen, 2020-09-23:
i use it this way :

```python
#!/usr/bin/env python3

# Using python package : https://pypi.org/project/pyFirmata/

from pyfirmata import Arduino, util

board = Arduino("/dev/ttyACM0")
it = util.Iterator(board)
it.start()

temp_sensor = board.get_pin("a:1:i")

def read_status_analog_pin(pin, counter: int = 10) -> float:
    pin.enable_reporting()

    read_list = []
    for _ in range(counter):
        read_list.append(pin.read())
        board.pass_time(1)

    pin.disable_reporting()

    result = round(sum(read_list) / len(read_list), 4)
    return result

if __name__ == "__main__":
    print(read_status_analog_pin(temp_sensor))

# ! # Read list
# 0.1697
# 0.1697
# 0.234
# 0.234
# 0.2664
# 0.2664
# 0.2661
# 0.2661
# 0.2679
# 0.2679
# 0.2694
# 0.2694
# 0.2693
# 0.2693
# 0.2683
# 0.2683
```

Kushal4092, 2020-09-24:
Try using resetting function once u connect to the board, after that use start function.

ozkansen, 2020-09-25:
How can I do this? I couldn't see anything about it in the existing code and documents.

Kushal4092, 2020-12-25:
Hi
I found a book on python coding for Arduino hope this helps

closed 2020-12-25 by Kushal4092