tino/pyFirmata #77: problem with basic usage code

asya-schtirlitz, 2018-11-03:
Hello, I am trying to light up a LED using your basic usage code from the Readme (but I use `board.digital[13]`, as it is the built-in LED for the Arduino Leonardo) and nothing happens. I also tried to light up LEDs connected to other digital pins. If I write a value into a digital pin and then read it, I successfully get the same value.

```python
from pyfirmata import Arduino, util
board = Arduino('/dev/tty.usbmodem141101')
board.digital[13].write(1)
```

Could you give me some advice?

I am using Arduino Leonardo and MacBook (Retina, 12-inch, 2017) with macOS 10.14, pyfirmata 1.0.3 and Python 3.6.5.

brettgoss, 2019-07-28:
Have you uploaded the StandardFirmata sketch to the Arduino already? See https://github.com/firmata/arduino#usage for more information on how to do that, or if you're using the Web Editor you can find it under Examples > StandardFirmata. Once you have it, upload it to the Arduino and then you should be able to use pyFirmata to communicate with it.

matiasandina, 2019-11-19:
Thank you! This should be more clear in the documentation, I skipped the StandardFirmata upload too!!