tino/pyFirmata #84: pyfirmata.py: DeprecationWarning

drumjoss, 2019-07-26:
Hi there!
With Python 3.6 I get this warning:
pyfirmata.py:185: DeprecationWarning: inspect.getargspec() is deprecated since Python 3.0, use inspect.signature() or inspect.getfullargspec() 
len_args = len(inspect.getargspec(func)[0])

I installed last week with pip.

wovano, 2022-11-26:
In Python 3.11 the `getargspec()` function has finally been removed and the pyFirmata library now gives an error:
```none
AttributeError: module 'inspect' has no attribute 'getargspec'. Did you mean: 'getargs'?
```

kiki-jpg, 2023-03-30:
could this be a solution ? 
#len_args = len(inspect.getargspec(func)[0])
sig = inspect.signature(func)
        params = sig.parameters
        len_args = len(params)

cyberexpertamit, 2023-07-19:
Replace len_args = len(inspect.getargspec(func)[0]) with len_args = len(inspect.getfullargspec(func)[0]

gnthibault, 2023-07-30:
I confirm from python 3.11 onwards, pyFirmata cannot be used as is.