tino/pyFirmata #121: ImportError

jsmwrench, 2022-11-20:
Hello.

After following the instructions (several times, same result), I keep getting the "ImportError" that states:
    Quote:  File "/home/Pi/Documents/LED_test.py", line 1, in <module>
                     from pfirmata import Arduino, util
               ImportError: cannot import name 'Arduino' from 'pyfirmata' (/home/Pi/Documents/pyfirmata.py)  :End Quote

What can I do to resolve this?

Thank you,
Ron.

Fat-Fox, 2022-11-20:
try: 

`from pyfirmata import Arduino, util`

you are missing the "y" in qour Quote. make also sure that the module is in your script directory or manipulate the search path accordingly.

jsmwrench, 2022-11-20:
I'm really new to Pi and Python programming, but I figured out that I had to find and copy the pyfirmata folder into the same folder as my sketch.  It's working now.  I have to assume this is like the Libraries in Arduino where all of the Libraries being used have to be in a pre-defined folder within the program.  Is this correct?  Are the Libraries for Python supposed to reside in "/pythonX.X/dist-packages"?  Should I then keep all of my sketches in the same folder?