tino/pyFirmata #122: inspect.getargs - Not available in Py 3.11 - Replace with inspect.getfullargspec

williamjcroke, 2022-12-06:
Replace "inspect.getargs" on line 185 of pyfirmata.py with "inspect.getfullargspec". 

Please update the repo code to this for Python 3.11 and later.

Halvhjearne, 2023-03-03:
this is literally only this line that needs to change, can you pls update the repo code?

DansDesigns, 2023-08-16:
+1 to the request

ChillarAnand, 2023-10-04:
@tino Is this repo actively maintained? I can send a PR for this.

groaking, 2023-11-26:
Commit [1f6b116](https://github.com/groaking/pyFirmata/commit/1f6b116b80172e70c7866d595120413078ae1222) has solved the bug. Line no. 185 now reads `inspect.getfullargspec` instead of the deprecated `inspect.getargs`.

However, the PIP package `pyFirmata` seems no longer developed since March 2019 ([source](https://pypi.org/project/pyFirmata/)). I would recommend switching to [**`pyFirmata2`**](https://pypi.org/project/pyFirmata2/). It works well with Python3.11 and is actively being developed. The syntax for both versions are also identical. You just need to replace `import pyfirmata as pf` with `import pyfirmata2 as pf` in you Python codes.

myselfgautham, 2024-04-14:
The Same Issues Has Occured On Python3.12