tino/pyFirmata #130: Pyfirmata not working but there are no errors

oanamoc, 2024-03-23:
I have written a code that should move some wheels on a robot. It works as expected in Arduino IDE. Original code:

```
#define M1INA 22
#define M1INB 23
#define M2INA 24
#define M2INB 25
#define PWM1 4
#define PWM2 5

void powerOffAllMotors()
{
digitalWrite(M1INA, LOW);
digitalWrite(M1INB, LOW);
digitalWrite(M2INA, LOW);
digitalWrite(M2INB, LOW);
}

void Forwards(int vel) //vel e procent
{
digitalWrite(M1INA, HIGH);
digitalWrite(M1INB, LOW);
digitalWrite(M2INA, HIGH);
digitalWrite(M2INB, LOW);
vel = vel/100*255;
analogWrite(PWM1,vel);
analogWrite(PWM2,vel);
}

void Backwards(int vel)
{ 
digitalWrite(M1INA, LOW);
digitalWrite(M1INB, HIGH);
digitalWrite(M2INA, LOW);
digitalWrite(M2INB, HIGH);
vel = vel/100*255;
analogWrite(PWM1,vel);
analogWrite(PWM2,vel);
}

void Right(int vel)
{
digitalWrite(M1INA, HIGH);
digitalWrite(M1INB, LOW);
digitalWrite(M2INA, HIGH);
digitalWrite(M2INB, LOW);
vel = vel/100*255;
analogWrite(PWM1,vel);
analogWrite(PWM2,vel/2); 
}

void Left(int vel)
{
digitalWrite(M1INA, HIGH);
digitalWrite(M1INB, LOW);
digitalWrite(M2INA, HIGH);
digitalWrite(M2INB, LOW);
vel = vel/100*255;
analogWrite(PWM1,vel/2);
analogWrite(PWM2,vel);  
}

void setup() {
pinMode(M1INA,OUTPUT);
pinMode(M1INB,OUTPUT);
pinMode(M2INA,OUTPUT);
pinMode(M2INB,OUTPUT);
pinMode(PWM1,OUTPUT);
pinMode(PWM2,OUTPUT);
powerOffAllMotors();
}

void loop() {
  Forwards(100);
  delay(2000);
  Backwards(100);
  delay(2000);
  Right(100);
  delay(2000);
  Left(100);
  delay(2000);
  powerOffAllMotors();
  delay(5000);
}
```
I translated the code to python with pyfirmata and it simpy doesn't work. There are no errors, but the wheels aren't turning. Here is the code:

```
# Import necessary libraries
import pyfirmata
import time

# Function to turn off all motors
def power_off_all_motors():
    M1INA.write(0)
    M1INB.write(0)
    M2INA.write(0)
    M2INB.write(0)

# Function to move forward
def forwards(vel):
    # Ensure motors are set to move forwards
    M1INA.write(1)
    M1INB.write(0)
    M2INA.write(1)
    M2INB.write(0)
    PWM1.write(vel)
    PWM2.write(vel)

# Function to move backwards
def backwards(vel):
    # Ensure motors are set to move backwards
    M1INA.write(0)
    M1INB.write(1)
    M2INA.write(0)
    M2INB.write(1)
    PWM1.write(vel)
    PWM2.write(vel)

# Function to turn right
def right(vel):
    # Ensure motors are set to move forwards
    M1INA.write(1)
    M1INB.write(0)
    M2INA.write(0)
    M2INB.write(1)
    PWM1.write(vel)
    PWM2.write(vel/2)

# Function to turn left
def left(vel):
    # Ensure motors are set to move forwards
    M1INA.write(0)
    M1INB.write(1)  # Changed for left turn
    M2INA.write(1)
    M2INB.write(0)
    # Convert velocity from percentage to PWM value and ensure it's within byte range
    v = max(0, min(int(vel / 100 * 255), 255))
    # Set different velocities for turning
    PWM1.write(vel/2) 
    PWM2.write(vel)

# Main code block
if __name__ == '__main__':
    # Setup communication with Arduino
    board = pyfirmata.ArduinoMega('/dev/tty.usbmodem1301')
    print("Communication Successfully started")
    
    # Initialize pins
    PWM1 = board.get_pin('d:4:p')
    PWM2 = board.get_pin('d:5:p')
    M1INA = board.get_pin('d:22:o')
    M1INB = board.get_pin('d:23:o')
    M2INA = board.get_pin('d:24:o')
    M2INB = board.get_pin('d:25:o')

    # Turn off all motors initially
    power_off_all_motors()

    # Main control loop
    # while True:
    forwards(100)
    time.sleep(6)
    print("done going forwards")
    # backwards(100)
    # time.sleep(2)
    # print("done going backwards")
    # right(100)
    # time.sleep(2)
    # print("done going right")
    # left(100)
    # time.sleep(2)
    # print("done going left")

```
I have uploaded the StandardFirmata code to the arduino from Arduino IDE so it can't be this. Any ideas?

myselfgautham, 2024-04-14:
**I Think This Should Fix It ->**

* Move All The Pins Variable And Board Variables And Connections To Top
* Add global followed by varname in Each Function

myselfgautham, 2024-04-14:
[Just An Attempt :atom:](https://gist.github.com/myselfgautham/f6e9179025304e37b0d139eef7430023)